Geonhee-LEE/mpc_ros #1: I want some help

HsiaoRay, 2019-05-02:
Hello, I find your project very attractive and I want to learn some knowledge about it. I see that the “zetabot-ros” is “private”, is this a  pack about a specific robot? And I wonder if the parts involved can be replaced with others(like turtlebot). Thank you for your help.

Geonhee-LEE, 2019-05-06:
Hello @HsiaoRay .

I'm sorry that "zetabot-ros" is a project with the company. So, I don't have qualification to give you some source. 

Why don't you use the [turtlebot3](http://emanual.robotis.com/docs/en/platform/turtlebot3/overview/) model as you commented?

The target is the differential wheeled mobile robot, because, you can use any turtlebot model.

Thanks for interesting.

closed 2019-05-09 by Geonhee-LEE

Geonhee-LEE, 2020-05-02:
@HsiaoRay  Hello, I recently uploaded the new branch which can use public model. You can use the melodic branch in this repo to use MPC. Thanks.