Geonhee-LEE/mpc_ros #3: Hello！Dear Friend! Is this project a plungin or a Complete project?

abc283646346, 2020-05-27:
If I use the Navigation stack of ROS, this project is a substitute of the dwa_local_planner. Is that right?

If it is a complete project, but I can't catkin_make this project. Is there something omitted with the operating steps?

Thank you!

abc283646346, 2020-05-27:
/home/wxd/catkin_ws_mpc/src/servingbot_ros/serving_bot/src/serving_bot.cpp:48:43: fatal error: find_object_2d/ObjectsStamped.h: 没有那个文件或目录

Geonhee-LEE, 2020-05-27:
Hi,

When I used it, It can be replaced with MPC in order to use Move_base. But
It has the some problems still such as obstacle avoidance.

Thanks for your interest.

Best regards.
Geonhee.

-- 

===============================================
Geonhee Lee
Graduate student
Department of Mechanical System Design Engineering
Seoul National University of Science & Technology
232 Gongneung-ro, Nowon-gu
<https://maps.google.com/?q=232+Gongneung-ro,+Nowon-gu+SEOUL,+KOREA&entry=gmail&source=g>
SEOUL, KOREA
<https://maps.google.com/?q=232+Gongneung-ro,+Nowon-gu+SEOUL,+KOREA&entry=gmail&source=g>
 (01811)

E-mail : gunhee6392@gmail.com

Homepage : https://geonhee-lee.github.io/geon.github.io/

GitHub: https://github.com/Geonhee-LEE

Tel(Cell phone): (+82)10-2888-2133

closed 2020-07-19 by Geonhee-LEE

ruficol, 2021-11-23:
> /home/wxd/catkin_ws_mpc/src/servingbot_ros/serving_bot/src/serving_bot.cpp:48:43: fatal error: find_object_2d/ObjectsStamped.h: 没有那个文件或目录

sudo apt-get install ros-kinetic-find-object-2d

Geonhee-LEE, 2021-11-23:
Hello, @ruficol.
Actually, you don't need to install find-object-2d package. But, you can install find-object-2d package on `http://wiki.ros.org/find_object_2d` in order to build successfully.

@CzJaewan could you update the servingbot package?

ruficol, 2021-11-24:
> Hello, @ruficol. Actually, you don't need to install find-object-2d package. But, you can install find-object-2d package on `http://wiki.ros.org/find_object_2d` in order to build successfully.
> 
> @CzJaewan could you update the servingbot package?
hi, Geonhee
i'm glad you replied to me. I have the same problem as abc283646346. After I installed find_object_2d, the error solved. 
but i have another error at /servingbot_ros/serving_bot/src/serving_bot.cpp: 63 fatal error
[image]
i can't solve this, so i commented this header file. After all this, i installed successfully.
my ubuntu version is 16.04.
ros: kinetic

i installed mpc_ros from kinetic branch, servingbot from NSCL.
the kinetic branch launch file does not have nav_gazebo.launch and ref_trajectory_tracking_gazebo.launch, so i copy this two launch files into my workspace, and yaml files, too.
i launch this: roslaunch mpc_ros nav_gazebo.launch
everything is fine, but i can't navigate, i send the "2D Nav Goal", the robot did't move at all. and my terminal mpc_node: process has died.
also i can't launch ref_trajectory_tracking_gazebo.launch, after launch this, my gazebo crashed. 

i'm sorry i have encountered so many problems. you can answer me when you have time.
so much thanks.