Geonhee-LEE/mpc_ros #5: The global status error

chanzw, 2020-09-07:
Excuse me! I have a question ,Follow the README.md, and I run the mpc_ros nav_gazebo.launch, the RVIZ has an error .
The global status error ,Fixed Frame [map] does not exist.Can you help me to solve it ? Thank you .

Geonhee-LEE, 2020-09-08:
Hello, Thanks for interesting. 
Did you install the map_server?
I guess it is the TF problem between map and odom.

chanzw, 2020-09-08:
> Hello @chanzw, Thanks for interesting.
> Did you install the map_server?
> I guess it is the TF problem between map and odom.

I have installed it. But it still has the same error.

closed 2020-09-08 by chanzw

Geonhee-LEE, 2020-09-08:
Please write your frame and load the map exactly .

chanzw, 2020-09-08:
> Please write your frame and load the map exactly .
OK! I try it.Thank you.

chanzw, 2020-09-08:
I can't solve it 
It always has a WARN

[ WARN] [1599573109.833110525, 98.097000000]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist. canTransform: source_frame base_link does not exist.. canTransform returned after 98.095 timeout was 0.1.

And  the RobotModel Status is error

Geonhee-LEE, 2020-09-08:
Hello, would you check the tf tree using 'rosrun rqt_tf_tree rqt_tf_tree' command on terminal? 
Because the tf structure has problem. And I wonder whether you has the robot model such as [servingbot](https://github.com/NSCL/servingbot_ros) or not.

chanzw, 2020-09-09:
> Hello, would you check the tf tree using 'rosrun rqt_tf_tree rqt_tf_tree' command on terminal?
> Because the tf structure has problem. And I wonder whether you has the robot model such as [servingbot](https://github.com/NSCL/servingbot_ros) or not.

Yes, I have checked it already.But it shows like that.Is it right?
[image]

And the rviz like this
[image]

I has download the servingbot model,when I catkin_make it ,it shows a error about simple_speech,I don't think it is the main reason,so I have delete the include "#include <simple_speech/script.h>".

And I have a husky robot.

Geonhee-LEE, 2020-09-09:
Hello.
You need to publish odom frame and connect between odom and map frame.

chanzw, 2020-09-09:
> Hello.
> You need to publish odom frame and connect between odom and map frame.

OK!Thank you very much.

chanzw, 2020-09-09:
Hello,I want to know the references for this NMPC,thanks!

Geonhee-LEE, 2020-09-09:
Hello, I strongly referred to https://github.com/Hypha-ROS/hypharos_minicar.
But it is based on Ackermann model

-- 

===============================================
Geonhee Lee
Graduate student
Department of Mechanical System Design Engineering
Seoul National University of Science & Technology
232 Gongneung-ro, Nowon-gu
<https://maps.google.com/?q=232+Gongneung-ro,+Nowon-gu+SEOUL,+KOREA&entry=gmail&source=g>
SEOUL, KOREA
<https://maps.google.com/?q=232+Gongneung-ro,+Nowon-gu+SEOUL,+KOREA&entry=gmail&source=g>
 (01811)

E-mail : gunhee6392@gmail.com

Homepage : https://geonhee-lee.github.io/geon.github.io/

GitHub: https://github.com/Geonhee-LEE

Tel(Cell phone): (+82)10-2888-2133

chanzw, 2020-09-14:
hello,When I set 2D Nav Goal,it will show that mpc_ref_path "Message contained invalid floating point values (nans or infs)" in rviz.AND in the terminal , it shows 
[ INFO] [1600048363.795110728, 1461.510000000]: Goal Received :goalCB!
coeffs : -nan
pow : 1
cte : -nan
etheta: 0, atan2(gy,gx): -0.94423, temp_theta:-0.94423
x_err:2.11219, y_err:-2.82514
-----------------------------------------------
cost_cte, etheta, velocity: -nan, 0, 2000
cost of actuators: -nan
cost of gap: -nan

******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit http://projects.coin-or.org/Ipopt
******************************************************************************

cppad-20180000.0 error from a known source:
yq = f.Forward(q, xq): a zero order Taylor coefficient is nan.
Corresponding independent variables vector was written to binary a file.
vector_size = 158
file_name = /tmp/fileGwXM0y
index = 0

Error detected by false result for
    if( CppAD::isnan( yq[ (q+1) * index + 0 ] )
at line 231 in the file 
    /home/czw/husky_ws/src/mpc_ros/mpc_ros/include/cppad/core/forward.hpp
nav_mpc: /home/czw/husky_ws/src/mpc_ros/mpc_ros/include/cppad/utility/error_handler.hpp:206: static void CppAD::ErrorHandler::Default(bool, int, const char*, const char*, const char*): Assertion `false' failed.
[nav_mpc-2] process has died [pid 16401, exit code -6, cmd /home/czw/husky_ws/devel/lib/mpc_ros/nav_mpc __name:=nav_mpc __log:=/home/czw/.ros/log/97b3b054-f62c-11ea-b2f4-9cd21efd7405/nav_mpc-2.log].
log file: /home/czw/.ros/log/97b3b054-f62c-11ea-b2f4-9cd21efd7405/nav_mpc-2*.log

Geonhee-LEE, 2020-09-14:
I think that is associated with IpOpt lib problem. Did you install the
IpOpt with right version?

-- 

===============================================
Geonhee Lee
Graduate student
Department of Mechanical System Design Engineering
Seoul National University of Science & Technology
232 Gongneung-ro, Nowon-gu
<https://maps.google.com/?q=232+Gongneung-ro,+Nowon-gu+SEOUL,+KOREA&entry=gmail&source=g>
SEOUL, KOREA
<https://maps.google.com/?q=232+Gongneung-ro,+Nowon-gu+SEOUL,+KOREA&entry=gmail&source=g>
 (01811)

E-mail : gunhee6392@gmail.com

Homepage : https://geonhee-lee.github.io/geon.github.io/

GitHub: https://github.com/Geonhee-LEE

Tel(Cell phone): (+82)10-2888-2133

chanzw, 2020-09-14:
> […](#)
> -- =============================================== Geonhee Lee Graduate student Department of Mechanical System Design Engineering Seoul National University of Science & Technology 232 [...]

I have install the Ipopt 3.12.7

chanzw, 2020-09-14:
> […](#)
> -- =============================================== Geonhee Lee Graduate student Department of Mechanical System Design Engineering Seoul National University of Science & Technology 232 [...]

I have installed the Ipopt 3.12.7

chanzw, 2020-09-14:
> […](#)
> -- =============================================== Geonhee Lee Graduate student Department of Mechanical System Design Engineering Seoul National University of Science & Technology 232 [...]

Could you share your Ipopt lib with me? My e-mail is cumt2czw@163.com.Thank you!

Geonhee-LEE, 2020-09-15:
@chanzw  Hello, I installed the Ipopt through https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/assets/document/ipopt_install/ipopt_x86_install_tutorial.md.