Geonhee-LEE/mpc_ros #7: An error is reported when the package is running, and the path cannot be followed

leigetiwen, 2020-12-17:
hello ,I ran this package and encountered some errors 。Terminal display this:
-----------------------------------------------
cost_cte, etheta, velocity: -nan, -nan, 10000
cost of actuators: -nan
cost of gap: -nan
------------ Total Cost(solution): 0------------

Geonhee-LEE, 2020-12-18:
@leigetiwen 
Hi, let me know the exact situation? that error seems like abnormal calculation b/w path and robot

leigetiwen, 2020-12-18:
@Geonhee-LEE I just publish a path and then report this error.so  I want to know what is required for a fixed path？

Geonhee-LEE, 2020-12-18:
@leigetiwen I made the python file [python file](https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/mpc_ros/script/mpc_trajectory_generation.py) to create fixed path. please refer to `generation_desired_path` function

leigetiwen, 2020-12-19:
Okay， thank you

closed 2020-12-19 by Geonhee-LEE