Geonhee-LEE/mpc_ros #9: Difference between mpc_ros global_planner and navigation stack global_planner?

HappySamuel, 2021-01-11:
Hi @Geonhee-LEE 

I have tried out your mpc_ros in the gazebo. I found that there's additional global_planner in the package. May i know what's the difference between your mpc_ros global_planner and navigation stack global_planner? Because i have written the highway costmap plugins. But it can only work with the navigation stack global_planner, so i couldn't get it to work with your mpc_ros global_planner.

Best,
Samuel

Geonhee-LEE, 2021-01-13:
@HappySamuel  
Hello, the reason why I implemented the `global_planner` is for integrating the ROS navigation stack.
Actually, `mpc_ros` is not local planner in ROS but independent model to use in ROS. Therefore, you can work with this package but you should consider ROS navigation stack to turn off other local planner.

HappySamuel, 2021-01-14:
Hi @Geonhee-LEE 

Thanks for the clarification. Is there any documents/guidelines that i can refer to for parameters tuning on the mpc_ros? 

Best,
Samuel

Geonhee-LEE, 2021-01-17:
Hi @HappySamuel. 
Sorry for that. Parameters tuning is up to your preference, but it may be easy to regulate it.

closed 2021-01-17 by Geonhee-LEE