Geonhee-LEE/mpc_ros #10: ref_trajectory_tracking_gazebo.launch doesn't launch Gazebo

mjestebang, 2021-01-13:
Hello Geonhee,
We solved the error we told you few days ago. It was due to some servingbot libraries when I was doing the catkin_make.
But, now, launching roslaunch mpc_ros ref_trajectory_tracking_gazebo.launch opens rviz, but does not open Gazebo. 
Error:   Could not find the 'robot' element in the xml fil at line 80 in /build/urdfdom-YMMa9X/urdfdom-1.0.0/urdf_parser/src/model.cpp

Thank you very much in advance.

Geonhee-LEE, 2021-01-14:
Hello, @mjestebang what kind of model you use?

mjestebang, 2021-01-14:
Hello, we are using your serving bot model.

Obtener Outlook para Android<https://aka.ms/ghei36>

mjestebang, 2021-01-14:
Hello, we are using your serving bot model

Geonhee-LEE, 2021-01-14:
Hello @mjestebang 
please close the #8 issue  if you solved. 

And  I wonder `servingbot_ros` was already built in your work space?  I guess the urdf file didn't upload when you launched. please check again.

mjestebang, 2021-01-17:
Yes, i think that's the problem,
I don't know if I have some mistake because the comands to launch the urdf I think are correct.
 I have this lines in the code related to the robot model.

    
    <param name="robot_description" command="$(find xacro)/xacro.py $(find servingbot_description)/urdf/servingbot.urdf.xacro" if="$(eval model == 'serving_bot')"/>

    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="world_name" value="$(find gazebo_ros)/launch/empty_world.launch"/>
        <arg name="paused" value="false"/>
        <arg name="use_sim_time" value="true"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="headless" value="false"/>
        <arg name="debug" value="false"/>
    </include>

    <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-urdf -model servingbot -x $(arg x_pos) -y $(arg y_pos) -z $(arg z_pos) -R $(arg roll) -P $(arg pitch) -Y $(arg yaw) -param robot_description" />
     
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
        <param name="publish_frequency" type="double" value="50.0" />
    </node>

Geonhee-LEE, 2021-01-18:
@mjestebang  Hi,
Why don't you change the name from `servingbot` to `robot` in urdf.xacro file and launch file?

mjestebang, 2021-01-19:
Hi, we have tried to change the name of the robot, but it doesn't make any change in the .launch.
Apparently, the RViz does read correctly de xml file (.urdf.xacro). 
We realize that the mistake is that RVIZ doesn't receive odometry. 

[image]

On the other hand, when we launch the node, we get this message in the terminal.
Error:   Could not find the 'robot' element in the xml file
         at line 80 in /build/urdfdom-YMMa9X/urdfdom-1.0.0/urdf_parser/src/model.cpp

Thanks you so much.
We are waiting your response.

Geonhee-LEE, 2021-01-19:
@mjestebang Hi, did you install `robot_state_publisher` package as well?

mjestebang, 2021-01-20:
Thanks you so much.
You've been a great help.

closed 2021-01-20 by mjestebang