Geonhee-LEE/mpc_ros #14: Motion model (DDMR, Ackermann)

Geonhee-LEE, 2021-01-18:
- [ ] Add and separate the Motion model (DDMR, Ackermann) for generalization

ingramator, 2021-02-08:
Hi @Geonhee-LEE did you make any internal progress on this? I am very interested in an Ackermann motion model and also a 4 Wheel Independent steer like model (holonomic like motion). 

Great work so far!

Geonhee-LEE, 2021-02-09:
@ingramator  Hi, thanks for interesting!
I planned to control various model in order to use model predictive control. The Ackermann model has many type, which include front steering and rear steering, etc. Our target model is front steering once, we want to extend it later. Would you recommend the model type you think with images or reference link?

ingramator, 2021-02-09:
@Geonhee-LEE This sounds very good! Definitely start with front steering ackermann (like a car). 

[image]
My team can potentially help with rear-stear ackermann and dual-ackermann. 

Here is a picture of 4 wheel independent steer:
[image]

[4 Wheel Independent Steer model kinematics.](https://digital-library.theiet.org/docserver/fulltext/joe/2015/1/JOE.2014.0241.pdf?expires=1612866414&id=id&accname=guest&checksum=A25F61DABC7C038A7B1B40517DE33F75)

An example of 1 wheel in action [[Youtube](https://www.youtube.com/watch?v=IUnAn8NEyjU)]. Let me know if you have any questions :+1:

Geonhee-LEE, 2021-02-09:
@ingramator  Great! Your model is attractive but it seems like more complex model than simple car model. I will try this model if will take a opportunity. Antyway, feel free to make a PRs including a new model if you are interested in `mpc_ros` improvements.

closed 2022-03-30 by Geonhee-LEE