Geonhee-LEE/mpc_ros #18: MPCPlannerConfig.h file is missing in local_planner_plugin branch

balamuruganky, 2021-01-29:
@Geonhee-LEE : Thanks for the great package and code refactoring. I was about to start porting MPC as local_planner_plugin as you are doing currently. I would like to integrate https://github.com/balamuruganky/path_planning, if you are interested. I request you to add the missing file in local_planner_plugin branch to proceed further. Thanks.

balamuruganky, 2021-01-29:
@Geonhee-LEE : I realized that, the work just started. I will do the integration with melodic branch for now. Thanks.

Geonhee-LEE, 2021-01-30:
@balamuruganky 
Hello, thanks for interesting this repo. 
I will develop the local_planner_plugin later to improve the modulity ans performance, ROS plugin formulation. Anyway, your path planning repo is attractive for me!! However, it is implemented in Python. Do you have plan to develop the path planning in C++? Or could you recommend the references? 

Thanks.

closed 2021-02-07 by Geonhee-LEE