Geonhee-LEE/mpc_ros #21: TF error

ZyHuang97, 2021-02-24:
Thanks for your code.
I tried to reimplement it, but the robot didn't move at all. 
It seems like a tf problem. Could you give me any advice?

Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist. canTransform: source_frame base_link does not exist.. canTransform returned after 98.075 timeout was 0.1.

Thanks and regards.

Geonhee-LEE, 2021-02-28:
HI, @ZyHuang97. Thanks for interesting this package. Did you install `map_server` of ros?

aakashs0ni, 2021-03-01:
Hi, @Geonhee-LEE. I have a similar issue. I have installed map_server. Furthermore, I have robotmodel erro in Rviz. It says No transform from [base_link, body_link, lidar_link...] to [map].

wangpeihu, 2021-05-14:
@Geonhee-LEE Hello,Thank you for the code.I have run the code ,but broke into some trouble.I have installed map_server and robot_state_publisher,but still make it.

wangpeihu, 2021-05-14:
[image]
[image]
[image]
[image]
[image]
[image]

wangpeihu, 2021-05-14:
[image]

Geonhee-LEE, 2021-05-15:
Hi @aakashs0ni where is the odom tf and odom topic? If you want to run, the odom tf and odom topic are necessary.

gft-ai, 2021-09-29:
Hi, I am facing the same issue. How and where do you publish those topics from? Do I need to write a separate script for them?

gft-ai, 2021-09-29:
> Hi, I am facing the same issue. How and where do you publish those topics from? Do I need to write a separate script for them?

Found the problem. I was missing

`ros-melodic-gazebo-plugins`

closed 2021-09-30 by Geonhee-LEE