Geonhee-LEE/mpc_ros #23: The Robot isn't moving

harderthan, 2021-05-13:
I launch the launch file (`ref_trajectory_tracking_gazebo.launch`).
But, the robot isn't moving.

[image]

I think this issue is same as #5.
How to connect /map frame with /odom frame?

[image]

Geonhee-LEE, 2021-05-15:
Hi, @harderthan.
When I tested with `ref_trajectory_tracking_gazebo.launch`, it works well like following:
- [image]

`ref_trajectory_tracking_gazebo.launch` don't need to connect between `map` and `odom`. As I think, your gazebo model don't include odom tf and odom. I referred the rqt_graph and rqt_tf_tree . 
- rqt_graph:
    - [image]
- rqt_tf_tree:
    - [image]

harderthan, 2021-05-20:
Which one of branch for this test which you referred upper?
I used the default branch (melodic). There are no links as `imu_link`, `caster_**` like your result on it.

I didn't yet figure out why gazebo not publihed `/odom` with tf.   
[image]

harderthan, 2021-05-20:
Hello,  @Geonhee-LEE . 
I found the solution to solve this issue.
There was other [repository](https://github.com/CzJaewan/servingbot) on `devel` branch. 
`robot_descritption` packages of `CzJaewan/servingbot` repository have a model included `gazebo.xacro`.
The robot model should have `controller` of gazebo plugins like `libgazebo_ros_diff_drive.so`.
It is charge of functions to publish /odom topic or subscribe /cmd_vel topic. 

Now, every things are working well. 
Thanks.

[image]

closed 2021-05-20 by harderthan

Simon-Leung, 2022-07-12:
> Hello, @Geonhee-LEE . I found the solution to solve this issue. There was other [repository](https://github.com/CzJaewan/servingbot) on `devel` branch. `robot_descritption` packages of [...]

i have the same problem, and i use `CzJaewan/servingbot` repository, but didn't work.
my log as:
```
===== Parameters =====
pub_twist_cmd: 1
debug_info: 0
delay_mode: 1
frequency: 0.1
mpc_steps: 20
mpc_ref_vel: 0.5
mpc_w_cte: 2000
mpc_w_etheta: 1000
mpc_max_angvel: 1

!! MPC Obj parameters updated !!
[ INFO] [1657616973.038310000]: Waiting for global path msgs ~
[ INFO] [1657616973.466635100]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1657616973.481899100]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Error:   Could not find the 'robot' element in the xml file
         at line 80 in /build/urdfdom-YMMa9X/urdfdom-1.0.0/urdf_parser/src/model.cpp
shared memfd open() failed: Function not implemented
shared memfd open() failed: Function not implemented
[ INFO] [1657616974.756086700]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1657616974.951515800, 0.100000000]: Physics dynamic reconfigure ready.
idx: 1
[ INFO] [1657616975.276573500, 0.180000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1657616975.277190700, 0.180000000]: Starting Laser Plugin (ns = /)
[ INFO] [1657616975.281489800, 0.180000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1657616975.296703400, 0.180000000]: Starting plugin DiffDrive(ns = //)
[ WARN] [1657616975.297022200, 0.180000000]: DiffDrive(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1657616975.301444000, 0.180000000]: DiffDrive(ns = //): <tf_prefix> =
[ WARN] [1657616975.301889700, 0.180000000]: GazeboRosDiffDrive Plugin (ns = ) missing <publishTf>, defaults to 1
[ INFO] [1657616975.305337600, 0.180000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1657616975.311722400, 0.180000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1657616975.334580800, 0.180000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1657616975.338057000, 0.180000000]: DiffDrive(ns = //): Advertise odom on odom
[ INFO] [1657616975.344281000, 0.181000000]: imu plugin missing <frameName>, defaults to <bodyName>
idx: 2
idx: 3
idx: 4
```