Geonhee-LEE/mpc_ros #33: How to make a simulation with dwa?

liunini123, 2022-02-22:
Hi, I want to compare the simulation results of MPC and DWA. 

I completed the simulation with MPC, like this:
[image]

But I don't know how to make a simulation with DWA, like the right results in this picture.
[image]

Geonhee-LEE, 2022-02-22:
Hi, @liunini123 
Actually, you should launch the `[dwa_local_planner](http://wiki.ros.org/dwa_local_planner?distro=noetic)` in order to use DWA .  This repo also supports [dummy global_planner plugin](https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/mpc_ros/global_planner_plugin.xml). Could you refer to above link?

liunini123, 2022-02-23:
I have downloaded the navigation package and add some codes in ref_trajectory_tracking_gazebo.launch.
[image]

But the robot didn't move. And the terminal shows like: 
[image]

Can you tell me in detail how to start move_base  in the ref_trajectory_tracking_gazebo.launch. And choose DWA as the local planner?

liunini123, 2022-02-23:
Another problem is when I run "roslaunch mpc_ros mpc_local_planner.launch  controller:=dwa", the robot always turn right, didn't follow the path.

https://user-images.githubusercontent.com/70358354/155269681-5c09cb39-a39c-4223-babe-d181ae51cc0f.mp4

Geonhee-LEE, 2022-02-24:
Hi, @liunini123 

Did you change global planner with `GeonPlanner`? In order to deliver path from `dwa_trajectory_generation.py` to global planner, you should use `GeonPlanner` not Global Planner

Geonhee-LEE, 2022-02-24:
> mpc_local_planner

To get nice DWA performance, parameter tuning is necessary. Why don't you try to change the dwa params?

liunini123, 2022-02-27:
> Hi, @liunini123
> 
> Did you change global planner with `GeonPlanner`? In order to deliver path from `dwa_trajectory_generation.py` to global planner, you should use `GeonPlanner` not Global Planner

I changed the global planner with 'GeonPlanner', then the robot can move. Sometimes,  the robot stop moving because path.cost_ < 0  in DWAPlannerROS. 
[image]
[image]

https://user-images.githubusercontent.com/70358354/155880373-18d075ea-3e69-4615-9d8d-8437ed57f72b.mp4

liunini123, 2022-03-01:
When I use dwa, global path is displayed in red and local path is displayed in blue in this video. The recorded path is far from global path sometimes. What should I do to make local path near to global path?

https://user-images.githubusercontent.com/70358354/156103323-3feefe12-4023-4a96-b953-ea4ddbfe4068.mp4

Geonhee-LEE, 2022-03-01:
> When I use dwa, global path is displayed in red and local path is displayed in blue in this video. The recorded path is far from global path sometimes. What should I do to make local path near to global path?
> 
>  dwa1.mp4

Hi, @liunini123. In order to follow the every poses of path, it should cut with particular interval as local plan. As I remind,  `_pathLength` of [global path planner plugin](https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/mpc_ros/src/global_planner/global_planner.cpp) corresponds that I mentioned above.

liunini123, 2022-03-01:
> 

Ok, Thank you! This video is played with dwa.

https://user-images.githubusercontent.com/70358354/156136220-50b83423-61c5-497c-9869-82973b93407a.mp4

Geonhee-LEE, 2022-03-01:
@liunini123, you're welcome!
Actually, DWA is not a path tracking method but collision avoidance method. Therefore, DWA can't not help avoiding performance decrease in a path tracking experiment. I suggest the pure pursuit and Stanley method can be a better control group about MPC.
Finally, please close with comment if you finished your issue.

closed 2022-03-02 by liunini123