Geonhee-LEE/mpc_ros #35: IPOPT

GithubMingEnter, 2022-08-01:
hello 
when I use mpc_ros local planner plugin, I find a error
```c++
src/mpc_ros/mpc_ros/include/cppad/utility/error_handler.hpp:206:
 static void CppAD::ErrorHandler::Default(bool, int, const char*, const char*, const char*): Assertion `false' failed.
```
please tell me how to handle it
thanks

Geonhee-LEE, 2022-08-04:
Hi, @GithubMingEnter.
I have never see that error. Did you follow the [installation guide](https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/assets/document/ipopt_install/ipopt_arm_install_tutorial.md)?

zhaohaowu, 2022-08-08:
> Hi, @GithubMingEnter. I have never see that error. Did you follow the [installation guide](https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/assets/document/ipopt_install/ipopt_arm_install_tutorial.md)?

Hi, I find the same error
`vector: index greater than or equal vector size
Error detected by false result for
    i < length_
at line 476 in the file 
    /home/zhw/duo/test/src/mpc_ros/mpc_ros/include/cppad/utility/vector.hpp
tracking_reference_trajectory: /home/zhw/duo/test/src/mpc_ros/mpc_ros/include/cppad/utility/error_handler.hpp:206: static void CppAD::ErrorHandler::Default(bool, int, const char*, const char*, const char*): Assertion `false' failed.
`
I have followed the installation guide, but I can't run `./get.Mumps` and `./get.Metis` because the website can't be opened.

zhaohaowu, 2022-08-08:
> Hi, @GithubMingEnter. I have never see that error. Did you follow the [installation guide](https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/assets/document/ipopt_install/ipopt_arm_install_tutorial.md)?

Hi, I find the same error
`
vector: index greater than or equal vector size Error detected by false result for i < length_ at line 476 in the file /home/zhw/duo/test/src/mpc_ros/mpc_ros/include/cppad/utility/vector.hpp tracking_reference_trajectory: /home/zhw/duo/test/src/mpc_ros/mpc_ros/include/cppad/utility/error_handler.hpp:206: static void CppAD::ErrorHandler::Default(bool, int, const char*, const char*, const char*): Assertion false' failed.
`
I have followed the installation guide, but I can't run.**/get.Mumps** and**./get.Metis** because the website can't be opened.

sumingzs11, 2022-08-11:
I have followed the installation guide, but I can't run./get.Mumps and**./get.Metis** because the website can't be opened

sumingzs11, 2022-08-11:
I have followed the installation guide, but I can't run./get.Mumps and**./get.Metis** because the website can't be opened
I have the same problem.

Geonhee-LEE, 2022-08-12:
@zhaohaowu  @sumingzs11  could you upload the images for installation process for coin-or?

sumingzs11, 2022-08-12:
Thanks for you answer. I have solved the problem. I think there is an error in Mumps and Metis. When we execute " ./get.Mumps", it ouccured an HTTP error. The method is to use the ipopt package concluding these and  then make and make install.

DeepDuke, 2022-08-15:
@I met this problem too. I also cannot run `./get.Mumps` and `./get.Metis` . @sumingzs11 how do you solve this?

sumingzs11, 2022-08-15:
@DeepDuke

sumingzs11, 2022-08-15:
@DeepDuke 
you can give me an e-mail address so that I can give you the package.

DeepDuke, 2022-08-15:
@sumingzs11 My email is czm.hit@gmail.com. Thank you!

sumingzs11, 2022-08-15:
@DeepDuke 
There may be some error because I have sent the package three times. Each time my mail has been returned, because it is recognized as spam.

DeepDuke, 2022-08-15:
@sumingzs11 Sorry, please send to my another email 11849041@mail.sustech.edu.cn

DeepDuke, 2022-08-15:
@sumingzs11 I tried your package, it works. Thank you!

closed 2022-08-15 by Geonhee-LEE

zhaohaowu, 2022-08-16:
> @DeepDuke you can give me an e-mail address so that I can give you the package.
Can you send me the package too. My Email address is 2356940823@qq.com. Thank you!

zrd1234, 2022-11-10:
> @DeepDuke you can give me an e-mail address so that I can give you the package.

Can you send me the package too. My Email address is 779258494@qq.com. Thank you!

slamlixiaodong, 2023-04-19:
Can you send me the package too. My Email address is [dxchinaedu@163.com](mailto:dxchinaedu@163.com). Thank you!

slamlixiaodong, 2023-04-19:
> > @DeepDuke you can give me an e-mail address so that I can give you the package.
> 
> Can you send me the package too. My Email address is [779258494@qq.com](mailto:779258494@qq.com). Thank you!

Can you send me the package too. My Email address is [dxchinaedu@163.com](mailto:dxchinaedu@163.com). Thank you!

Raychan382, 2025-01-23:
@DeepDuke Can you send me the package too. My Email address is [1654145253@qq.com](mailto:1654145253@qq.com). Thank you!