Geonhee-LEE/mpc_ros #36: process has died

zrd1234, 2022-12-12:
Hello, I found that when the robot started tracking, it would report an error. What is the reason? Thank you very much.
[image]

Geonhee-LEE, 2022-12-15:
@zrd1234 Would you explain for more detailed information?? For examples, launch file and dependencies.

zrd1234, 2022-12-15:
> 嗨，@zrd1234。你能解释更详细的信息吗？例如，执行的启动文件和依赖项。

After I compile the program in the NX trajectory_tracking_gazebo.launch file, everything else runs, but when the model is going to move, the process is reported to die.
When I run roslaunch mpc_ros nav_gazebo.launch, everything works fine, but when I give it a target point, the process dies.
Thank you for your reply. Thank you.

zrd1234, 2022-12-15:
> Hi, @zrd1234. Would you explain for more detailed information?? For examples, executed launch file and dependencies.

After I compile the program in the NX trajectory_tracking_gazebo.launch file, everything else runs, but when the model is going to move, the process is reported to die.
When I run roslaunch mpc_ros nav_gazebo.launch, everything works fine, but when I give it a target point, the process dies.
Thank you for your reply. Thank you.

zrd1234, 2022-12-27:
Hello, when you have time, can you answer my question? I'm a little worried. Thank you very much.

Geonhee-LEE, 2022-12-28:
Hi, @zrd1234. 
In order to send the target goal via RViz(2D Goal Arrow), you need to replace the stand-along node with the move_base plugin such as DWA, etc. I tried to construct the dummy move_base plugin, but it have some limitations.
If you want to send the target goal, you can refer [mpc_plugin.xml](https://github.com/Geonhee-LEE/mpc_ros/blob/local_planner_plugin/mpc_ros/mpc_plugin.xml).

zrd1234, 2022-12-28:
Hello, is it possible that I run ref under the x86 architecture because of the architecture problem ref_ trajectory_ tracking_gazebo.launch file can succeed.
But when I run under the arm architecture, an error will be reported.
I don't understand what you mean by reference.
Thank you for your reply.

Geonhee-LEE, 2022-12-28:
> Hello, is it possible that I run ref under the x86 architecture because of the architecture problem ref_ trajectory_ tracking_gazebo.launch file can succeed. But when I run under the arm [...]

I sorry to that, but I haven't never tried to launch `mpc_ros` on arm architecture.
I checked the error log you uploaded: "[failed to path generation](https://github.com/Geonhee-LEE/mpc_ros/blob/5b6985e1abd9c95f49ace644729bae83484f868b/mpc_ros/src/navMPCNode.cpp#L300)". The meaning is that it can't make the path since too short distance.
Could you check the "mpc_reference" topic visualization on Rviz and decrease the "path_length" param value?

zrd1234, 2022-12-28:
> > 您好，由于体系结构问题，我是否有可能在x86架构下运行ref ref_ trajectory_ tracking_gazebo.launch文件可以成功。但是当我在手臂架构下运行时，会报告错误。我不明白你所说的参考是什么意思。感谢您的回复。
> 
> 对此我很抱歉，但我从未尝试过在 arm 架构上启动。我检查了您上传的错误日志：“[无法生成路径](https://github.com/Geonhee-LEE/mpc_ros/blob/5b6985e1abd9c95f49ace644729bae83484f868b/mpc_ros/src/navMPCNode.cpp#L300)”。意思是它不能走路，因为距离太短。您能否检查 Rviz 上的“mpc_reference”主题可视化并减少“path_length”参数值？`mpc_ros`
Thank you for your reply. I will try it and ask again if there are any questions. Thank you very much for replying to me so quickly. Thank you very much.

Geonhee-LEE, 2023-01-02:
If you solved this issue, please close the issue.

zrd1234, 2023-01-03:
> I sorry to that, but I haven't never tried to launch `mpc_ros` on arm architecture. I checked the error log you uploaded: "[failed to path [...]

Hello, I tried to reduce the path_length, but it still didn't work. The error is as follows.
[image]

Geonhee-LEE, 2023-01-05:
> > If you solved this issue, please close the issue.
> 
> Hello, could you please help me check the problem when you have time? Thank you very much. I really need this method to apply to my robot.

Hi, it is hard to guess what reason occurs according to screenshot. Could you add some logs on c++ files and check again which code has the problem?

zrd1234, 2023-01-05:
> Hi, it is hard to guess what reason occurs according to screenshot. Could you add some logs on c++ files and check again which code has the problem?

Hello, may I ask how to operate, this aspect is not very clear.

zrd1234, 2023-01-09:
> Hi, @zrd1234 . you can try to add some codes such as ROS_INFO, ROS_WARN on c++ file which you launched and re-launch after catkin_make build.

Hello, run ref_trajectory_tracking_gazebo. Launch, whether MPC. CPP and trackRefTraijnode CPP file to run.
rosinfo, these display messages, just to see which one will crash the program?

zrd1234, 2023-01-31:
Hello, I've been trying all this time, but I still can't solve it. Then I changed the mpc to pure_pursuit, but I couldn't move it, as shown in the screenshot below.
[image]
[image]

zrd1234, 2023-02-02:
Hello, could you please help me check the problem when you have time? Thank you very much. I really need this method to apply to my robot.

Geonhee-LEE, 2023-02-02:
Hi, @zrd1234 . Would you attach the rqt_graph?? I can't seize your situation exactly. Please upload it. 
The later screen-capture image show the "map" passed to ... that means didn't realize the total relationship for TF. It can't be seem to be `mpc_ros` problem I think

zrd1234, 2023-02-02:
> Hi, @zrd1234 . Would you attach the rqt_graph?? I can't seize your situation exactly. Please upload it. The later screen-capture image show the "map" passed to ... that means didn't realize the total relationship for TF. It can't be seem to be `mpc_ros` problem I think

Hello, I re-downloaded and ran the code.
I'm running a roslaunch mpc_ros ref_trajectory_tracking_gazebo. Launch
The situation is as follows:
[image]
[image]

Resister1, 2023-02-14:
方便加个微信吗？想请教一下你

Resister1, 2023-02-14:
> > > Hi, @zrd1234. Would you explain for more detailed information?? For examples, executed launch file and dependencies.
> > 
> > 
> > After I compile the program in the NX trajectory_tracking_gazebo.launch file, everything else runs, but when the model is going to move, the process is reported to die. When I run roslaunch mpc_ros nav_gazebo.launch, everything works fine, but when I give it a target point, the process dies. Thank you for your reply. Thank you.
> 
> Hello, I have the same question, have you solved this issue?

方便加个微信吗？想请教一下你

Geonhee-LEE, 2023-02-14:
Hi, @zrd1234 and @Resister1. Could you check the issue(#37)? I just updated the melodic branch for solving setPlan bug.

zrd1234, 2023-02-15:
> 方便加个微信吗？想请教一下你

可以，18563266055

zrd1234, 2023-02-15:
> Hi, @zrd1234 and @Resister1. Could you check the issue(#37)? I just updated the melodic branch for solving setPlan bug.
Hello, still can't solve this problem. There's something wrong with the generating path. Can you help me? I need this program very much.
[image]
[image]

zrd1234, 2023-02-15:
> Hi, @zrd1234 and @Resister1. Could you check the issue(#37)? I just updated the melodic branch for solving setPlan bug.

Hi, I've detected a problem. In the trackRefTraj.cpp file, around line 377: solve the problem, there is an error in this place, which leads to the process die. Do you have any suggestions?
[image]

Resister1, 2023-02-15:
> 

> Hi, @zrd1234 and @Resister1. Could you check the issue(#37)? I just updated the melodic branch for solving setPlan bug.

well, I just updated the melodic branch but it doesn't work. Also, I want to ask if this code can adapt to the ROS-noetic/Ubuntu 20.04.

Geonhee-LEE, 2023-02-15:
> Hi, I've detected a problem. In the trackRefTraj.cpp file, around line 377: solve the problem, there is an error in this place, which leads to the process die. Do you have any suggestions? [image]

It seems like Ipopt solver problem. Did you install coin-or Ipopt completely?

zrd1234, 2023-02-16:
> t seems like Ipopt solver problem. Did you install coin-or Ipopt completely?

I have only installed ipopt, do I need to install anything else?
When I run nav_gazebo.launch, I switch mpc to pursuit.

Resister1, 2023-02-16:
Dear Sir ,I would like to ask that when I use the global planner you designed as a plugin to the "move_base" of the nav_gazebo.launch file, if I give a target destination in Rviz, it doesn't produce a global path after I choose the corresponding the global planner topic. I am looking forward to your prompt reply. Thank you

Resister1, 2023-02-16:
Also, I would like to ask that what is the algorithm type in your global planner.

closed 2023-03-13 by zrd1234