Geonhee-LEE/mpc_ros #39: ipopt install error

sunki0220, 2023-02-16:
안녕하세요.
ipopt 설치과정중에 ./get.Mumps 이후로 넘어가지가 않습니다.
해결방법이 있을까요?
[image]

Geonhee-LEE, 2023-02-16:
안녕하세요, 해당 에러를 보지 못했었습니다. 혹시 의존성 및 CPU에 맞게 설치를 진행하셨을까요?

sunki0220, 2023-02-16:
확인해보니 64비트는 다른 방법으로 설치해야하나보네요!

sunki0220, 2023-02-16:
현재 ubuntu 18.04 버전이고 64bit입니다.
혹시 의존성설치를 올려주신 방법처럼 간편하게 진행하는 방법이 없을까요?
https://coin-or.github.io/Ipopt/INSTALL.html#INT64_BUILD     여기 링크대로 따라해보았는데 잘 안되네요.....

Geonhee-LEE, 2023-02-16:
https://github.com/Geonhee-LEE/mpc_ros/blob/melodic/assets/document/ipopt_install/ipopt_x86_install_tutorial.md 을 따라하셨는데, 안되는건가요?

sunki0220, 2023-02-16:
네 그렇습니다 ㅠ

yuguanfeng, 2023-04-18:
I found the reason is the website for downloading tar remotely is no longer useful.

Then I solved it by download MUMPS_4.10.0.tar.gz manually, put it in Mump fold and commented out the code for the remote download in get.Mumps which is `$wgetcmd http://mumps.enseeiht.fr/...`

Here is the corresponding version of MUMPS [https://github.com/fight-ldr/ipopt_install](url)

wcpcp, 2023-05-06:
do you have http://glaros.dtc.umn.edu/gkhome/fetch/sw/metis/OLD/metis-4.0.3.tar.gz,i cannot find the wetsite

yuguanfeng, 2023-05-06:
> do you have http://glaros.dtc.umn.edu/gkhome/fetch/sw/metis/OLD/metis-4.0.3.tar.gz,i cannot find the wetsite

i also find one in this website [AN304/metis-4.0.3.tar.gz at master · Hartorn/AN304 (github.com)](https://github.com/Hartorn/AN304/blob/master/metis-4.0.3.tar.gz)