Geonhee-LEE/mpc_ros #40: Replacement solver

zrd1234, 2023-03-13:
Hello, the ipopt solver seems to be unable to work under the arm architecture. May I ask what part of work should be done if you want to change the solution? Thank you.

Geonhee-LEE, 2023-03-15:
I recommend the optimization solver with CasADi, ACADOS in order to solve the MPC problem .

zrd1234, 2023-03-15:
Hello, do I just need to replace trackRefTraj.cpp with ipopt?

Geonhee-LEE, 2023-03-16:
@zrd1234 I think it is not easy to replace it. Because each solver's interface may be different such as Eigen, Vector, self-structured.

zrd1234, 2023-03-16:
> @zrd1234 I think it is not easy to replace it. Because each solver's interface may be different such as Eigen, Vector, self-structured.

Hello, I'm really sorry. If you are free, could you upload and modify the code of the solver? Thank you very much.

closed 2023-03-16 by Geonhee-LEE