Geonhee-LEE/mpc_ros #41: project

TWSTYPH, 2023-03-23:


TWSTYPH, 2023-03-23:
[
[image]
](url)
Error detected by false result for
    i < length_
at line 476 in the file 
    /home/yph/mpc_car/src/mpc_ros/mpc_ros/include/cppad/utility/vector.hpp
tracking_reference_trajectory: /home/yph/mpc_car/src/mpc_ros/mpc_ros/include/cppad/utility/error_handler.hpp:206: static void CppAD::ErrorHandler::Default(bool, int, const char*, const char*, const char*): Assertion `false' failed.
[MPC_tracking-6] process has died [pid 38533, exit code -6, cmd /home/yph/mpc_car/devel/lib/mpc_ros/tracking_reference_trajectory __name:=MPC_tracking __log:=/home/yph/.ros/log/b60f168c-c975-11ed-b5bc-000c295cfaf0/MPC_tracking-6.log].
log file: /home/yph/.ros/log/b60f168c-c975-11ed-b5bc-000c295cfaf0/MPC_tracking-6*.log

May I ask why there is always such a problem when mpc is running?

zrd1234, 2023-04-06:
> [ [image] ](url)在文件 /home/yph/mpc_car/src/mpc_ros/mpc_ros/include/cppad/utility/vector.hpp 的第 476 行< length_的错误结果检测到错误 tracking_reference_trajectory： /home/yph/mpc_car/src/mpc_ros/mpc_ros/include/cppad/utility/error_handler.hpp：206： static void CppAD：：ErrorHandler：:D efault（bool， int， const char*， const char*， const char*， const char*）：断言“false”失败。 [MPC_tracking-6] 进程已死亡 [PID 38533，退出代码 -6，cmd /home/yph/mpc_car/devel/lib/mpc_ros/tracking_reference_trajectory __name：=MPC_tracking __log：=/home/yph/.ros/log/b60f168c-c975-11ed-b5bc-000c295cfaf0/MPC_tracking-6.log]。日志文件： /home/yph/.ros/log/b60f168c-c975-11ed-b5bc-000c295cfaf0/MPC_tracking-6*.log
> 
> 请问为什么在mpc运行时总是有这样的问题？

Hello, I also encountered this problem, may I ask if you have solved it

Geonhee-LEE, 2023-04-18:
Hi, @TWSTYPH @liujiapeng888 
Would you share what the kind of command you launch and upload the Rviz display?

wanpeipei, 2024-10-16:
I have encountered this error with you before, and you need to reinstall Ipopt