Geonhee-LEE/mpc_ros #43: error

zrd1234, 2023-04-10:
Hello, when I run our program, I get the following error:
[image]
How can I solve it? Thank you.

haloworld-C, 2023-06-13:
@zrd1234 I think it's ipopt installation issue, the solver used by ipopt was not installed correctly. BTW, I make a docker named [guangaltman/mpc_demo:latest] after running the demo successfully, you can mount the repo code under ~/catkin_ws/src and run catkin_make to test code in this repo

zrd1234, 2023-06-19:
> 我认为这是ipopt安装问题，ipopt使用的求解器未正确安装。顺便说一句，我在成功运行演示后制作了一个名为 [guangaltman/mpc_demo：latest] 的 docker，您可以在 docker 中挂载 ~/catkin_ws/src 下的存储库代码并运行 catkin_make 以测试此存储库中的代码。

Thank you for your reply. I have solved this problem, which is only the problem of solver in arm architecture.

closed 2023-06-19 by zrd1234