Geonhee-LEE/mpc_ros #46: What is the meaning of d and η in the path tracking model? Can you explain it? Thank you!!!

MandyZhang4869, 2024-01-21:
What is the meaning of d and η in the path tracking model? Can you explain it? Thank you!!!

MandyZhang4869, 2024-01-21:
and  what is the meaning of θt*  ？

Geonhee-LEE, 2024-01-22:
Hi, @MandyZhang4869 
Did you read the paper "Path Tracking with Nonlinear Model Predictive Control or Differential Drive Wheeled Robot" written by Korean?
In this paper, the "d" term means the lateral error b/w robot's position and reference path's position and the "η" term means the angle error b/w robot's orientation and reference path's orientation. Finally, the "θt" term means time variable robot's orientation.

Thanks.

MandyZhang4869, 2024-01-22:
Having just read your paper, I still have a few questions. (I am
relatively new to this field and still absorbing a lot of information, so
my questions might be numerous and my understanding of your paper might be
somewhat skewed.)
1. According to your interpretation, should d and η in the figures be
reversed?   Additionally, I am a bit unsure about which points the error
refers to. Is it the error between the point on the reference line at time
t and the vehicle center B at time t?
2. Based on the graphs presented in your video, is the green curve the
reference trajectory for y given by the polynomial fitting curve g(xt)?
3. Is xt in g(xt) the current position of the vehicle measured at time t?
But if so, I am puzzled about what g(xt) is. It doesn't make sense to
directly substitute the current position into the fitting curve of the
reference trajectory, right?
4. I am not quite clear about the right-hand side of the equation for dt+1.
Especially what does vt·sinη·dt mean?
Looking forward to your reply！！！Thank you very much！！！

[image: 1705902718191.png]